golemparts/rppal #4: add support for interrupts in addition to current polling for GPIO input

vandenoever, 2018-04-26:
To handle signals on GPIO pins, a function to add a callback to a GPIO pin state is needed.

Polling can miss events and is possibly CPU intensive.

https://sourceforge.net/p/raspberry-gpio-python/wiki/Inputs/ has a discussion on the difference and explanation of Python functions to monitor GPIO inputs.

golemparts, 2018-04-26:
Thanks for the suggestion, and the link. I agree that adding support for GPIO interrupts would be nice to have.

I've briefly looked into adding interrupts before, but only came across sysfs-based solutions. While I do have direct access to the relevant registers (GPEDSn), there doesn't seem to be a way to specify an ISR vector table, and polling the registers ends up causing the same issue you pointed out.

Regardless, it's still worth it to use the sysfs interface just for interrupts, and I'll add it to the to do list. I'm open to suggestions for a better solution though.

vandenoever, 2018-04-26:
Perhaps you could use epoll.

https://github.com/rust-embedded/rust-sysfs-gpio/blob/master/src/lib.rs#L554

golemparts, 2018-04-26:
Yeah, using epoll to wait for sysfs changes is the plan. I would prefer more direct access though, but that doesn't currently seem possible in userspace.

golemparts, 2018-05-01:
Both synchronous and asynchronous interrupts have been added, and will be available as part of the upcoming 0.5.0 release.

closed 2018-05-01 by golemparts

braincore, 2018-08-03:
Do you have plans to support level-triggered interrupts? As far as I can tell, only edge triggered is supported.

golemparts, 2018-08-03:
@braincore the interrupt code in the version of rppal currently in development uses the new /dev/gpiochipN interface through epoll, rather than the outdated sysfs interface used in previous versions. Since I don't specifically set the `EPOLLET` flag, in theory epoll should already be configured for level-triggered interrupts, but in reality it behaves like an edge trigger.

The BCM283x registers do have options to configure either level- or edge-triggered interrupts, but it's possible the userspace drivers don't support those. I'll look into what the drivers offer, and if it would make sense to build a level-triggered option in software instead.

commit 2204e781 (2018-05-01): Update changelog with new interrupt features

Closes #4