golemparts/rppal #8: Replace GPIO sysfs interface for interrupts with GPIO character device

golemparts, 2018-07-04:
The GPIO sysfs interface (`/sys/class/gpio`) has been deprecated, with the relatively new GPIO character device (`/dev/gpiochipN`) interface offered as a replacement. Since GPIO sysfs is currently used for interrupts, a migration to the GPIO cdev interface needs to happen before GPIO sysfs is removed (after 2020).

As an added benefit, this might help as a workaround for a bug in Fedora where GPIO sysfs doesn't properly handle interrupts. (Still needs to be verified. A [proof of concept](https://github.com/golemparts/rppal/issues/7#issuecomment-402257815) is available for testing) @drahnr @KuabeM

golemparts, 2018-07-07:
All of the existing interrupt methods now use the new GPIO character device interface. On Raspbian they're working as intended.

KuabeM, 2018-07-08:
As well as on Fedora - thank you, great job!

closed 2018-08-14 by golemparts