golemparts/rppal #9: Guidance on singleton use

denizs, 2018-08-01:
Hey there,
I just stumbled upon your project and was wondering whether you could provide some guidance wrt implementing GPIO as a singleton. I've already tried to use `lazy_static` and instantiating it as a `static ref`, both approaches do not work.
Best,
D

golemparts, 2018-08-01:
What issue are you running into? While global variables are discouraged, if you do need to use one, you can do the following:

```rust
use rppal::gpio::Gpio;

lazy_static! {
    static ref GPIO: Mutex<Gpio> = Mutex::new(Gpio::new().unwrap());
}
```

After which you can access the Gpio instance from any thread by unlocking the mutex with:

```rust
let mut gpio = GPIO.lock().unwrap();
```

denizs, 2018-08-01:
Ok it seems like this wasn't rppal related. Sorry about that!
In the meantime, I moved a single instance of `Gpio` alongside a `sync_channel` `Receiver` into a separate thread, passing the `SyncSender` instance to threads, which need to have access to gpio pins.

closed 2018-08-01 by denizs