golemparts/rppal #10: Research level-triggered interrupts

golemparts, 2018-08-03:
Check if the new `/dev/gpiochipN` interface with epoll allows level-triggered interrupts as well as the current edge-triggered interrupts. Initial research seems to indicate epoll uses edge triggers regardless of whether or not `EPOLLET` is set. If the drivers don't support it, see if it makes sense to add level-triggered interrupts in software.

braincore, 2018-08-03:
Thank you for opening the issue. I can confirm that I don't see level triggers even when omitting a call to `set_interrupt()`.

Do you have a "software" workaround for level triggering that avoids the obvious race condition?

```
check_gpio_value()  // returns 0
// <-- gpio is brought high
poll_gpio() // no edge trigger since gpio is already high
```

golemparts, 2018-08-04:
Are you using the latest released version, or the development version in the master branch in the repo? They might behave differently. I recommend switching to the master branch for now. I'll likely release the next update in the coming week, without the uart module, just so I can get the interrupt changes and support for hardware PWM channels out the door.

With the master branch version: If you're using synchronous/blocking interrupts, after configuring an interrupt with `set_interrupt`, if an edge trigger occurs in between two calls to `poll_interrupt` or `poll_interrupts`, the second call should return immediately with the relevant pin/level data, provided the `reset` parameter is set to `false`. After that, it won't trigger again until another edge trigger occurs. You should theoretically be able to turn that into something similar to level triggering by setting triggers for both rising and falling edges, and saving that state in your own code. If you don't see that happening with the current master branch, there may be a problem somewhere.

golemparts, 2019-02-27:
In an effort to better organize feature requests, and considering this request has been open for 6 months and is unlikely to be addressed within the next few weeks, I'm closing this for now and have added level-triggered interrupts to the Research section of [feature requests](https://github.com/golemparts/rppal/projects/10). If it moves to the Planned section and further discussion is warranted, this issue can be reopened.

The underlying Linux drivers currently don't support this even though the BCM283x's registers do allow interrupts to be configured for level-triggering. Implementing a software-based solution in rppal's interrupt routines could be an option, provided it doesn't interfere with the edge-triggered implementation.

As mentioned, until rppal includes support for level-triggered interrupts, you can use the current implementation for something similar to level-triggering by relying on rppal's ability to cache triggers that occur in between calls to `poll_interrupt()`, which prevents race conditions when the pin state changes in between `read()` and `poll_interrupt()`.

closed 2019-02-27 by golemparts