golemparts/rppal #19: SetInterrupt for input with pulldown resistor

rahul-thakoor, 2019-02-02:
Hey, I am trying to set a synchronous interrupt(trigger is Rising Edge) on a GPIO which has been configured as input with pulldown resistor. Assuming I am connecting a push button, with one side connected to GPIO input and the other to 3.3V. The expected behaviour is that the interrupt fires when button is pressed. However, this only happens if I connect other side of button to ground instead of 3.3v. 
For context, I am trying to port this:
https://gpiozero.readthedocs.io/en/stable/api_input.html#button
How can I resolve this please?
Thanks

golemparts, 2019-02-02:
I just tried this myself, and everything seems to work as expected. I connected one side of a button to GPIO 23, and the other side to 3.3 V. When I press the button, the interrupt triggers. Perhaps you accidentally chose two pins on the tactile switch that are connected together? For reference, this is the code I used:

```rust
use std::error::Error;

use rppal::gpio::{Gpio, Trigger};

const GPIO_BTN: u8 = 23;

fn main() -> Result<(), Box<dyn Error>> {
    let mut pin = Gpio::new()?.get(GPIO_BTN)?.into_input_pulldown();

    pin.set_interrupt(Trigger::RisingEdge)?;

    println!("Waiting for rising edge trigger...");
    pin.poll_interrupt(true, None)?;

    println!("Triggered!");

    Ok(())
}
```

rahul-thakoor, 2019-02-03:
Hey there! My bad... It did work as described for other GPIO pins. I was having problems because I was trying it on BCM 2 and BCM 3 which are i2c SDA and SCL respectively. Those pins have fixed 1.8 kohms pull-up to 3.3v. I believe that was causing the issue. 
Thanks!

golemparts, 2019-02-03:
Hah, that does indeed explain the strange behavior. Glad to hear you figured out the cause!

closed 2019-02-03 by golemparts