golemparts/rppal #21: Error on generating docs when used as a dependency

rahul-thakoor, 2019-02-24:
Hey @golemparts! I am using rppal 0.11 in https://github.com/rahul-thakoor/rust_gpiozero which works fine but when I am generating the docs using `cargo doc --no-deps` I get the following errors:

````   
 Checking rppal v0.11.0
error[E0308]: mismatched types
   --> /home/vagrant/.cargo/registry/src/github.com-1ecc6299db9ec823/rppal-0.11.0/src/gpio/soft_pwm.rs:232:17
    |
232 |         tv_sec: (ns / 1_000_000_000) as i32,
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected i64, found i32
help: you can cast an `i32` to `i64`, which will sign-extend the source value
    |
232 |         tv_sec: ((ns / 1_000_000_000) as i32).into(),
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0308]: mismatched types
   --> /home/vagrant/.cargo/registry/src/github.com-1ecc6299db9ec823/rppal-0.11.0/src/gpio/soft_pwm.rs:233:18
    |
233 |         tv_nsec: (ns % 1_000_000_000) as i32,
    |                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected i64, found i32
help: you can cast an `i32` to `i64`, which will sign-extend the source value
    |
233 |         tv_nsec: ((ns % 1_000_000_000) as i32).into(),
    |                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to 2 previous errors

````

Any suggestions? Thanks

golemparts, 2019-02-24:
The error is caused by a difference in data types used by the `timespec` fields for 32-bit and 64-bit linux. I fixed it on the master branch a few days ago, and it's scheduled for release as 0.11.1 within the next week or so (along with uart peripheral support).

Currently only a 32-bit version of Raspbian is available, so you would normally not encounter this error on a Pi with a default installation. I discovered it myself when I accidentally compiled rppal 0.11.0 on 64-bit linux by not specifying a target for Travis CI.

You can circumvent the error until 0.11.1 is released either by setting the build target to `armv7-unknown-linux-gnueabihf`, or temporarily depending on the master branch until the next release is available.

Let me know if that worked for you.

golemparts, 2019-02-24:
Actually, I think I'll accelerate the release of 0.11.1 now that someone has encountered the error, and save `Uart` for 0.11.2. I'll have time to prepare 0.11.1 for release later today, so you should see an update within the next 24 hours.

jens1o, 2019-02-24:
@golemparts Thanks for the reconsideration. I just wanted to ask whether you could file the release a little bit sooner.

golemparts, 2019-02-24:
@rahul-thakoor @jens1o rppal 0.11.1 is now available on crates.io, containing a fix for this error. `cargo update` should be able to grab it for you.

closed 2019-02-24 by golemparts