golemparts/rppal #22: How to control a servo?

jens1o, 2019-02-25:
Hi there,

I want to attach a servo to my Raspberry Pi 3 B+. I think they are controlled using PWM. Could you give me a hint how I could use the libraries's PWM module to control it?

golemparts, 2019-02-25:
Sure! I think the examples in the `examples` directory can probably help you out.

Hardware PWM servo example: https://github.com/golemparts/rppal/blob/master/examples/pwm_servo.rs
Software-based PWM servo example: https://github.com/golemparts/rppal/blob/master/examples/gpio_servo_softpwm.rs

While both hardware and software-based PWM will work, software-based PWM is inherently inaccurate on a multi-threaded OS due to scheduling/preemption. If you only need up to two PWM channels, I recommend just sticking with hardware PWM.

Be careful when hooking up a servo to your Pi. If you power your servo directly from the Pi's 5 V output, a current spike caused by the servo starting up or stalling could best case temporarily blow the polyfuse, worst case permanently damage your Pi. Most 9g micro servos are probably safe enough if you only connect a single servo, but definitely check its maximum ratings in the datasheet, or use an external power supply (connect the grounds together) to be safe. You'll also need to connect the signal line to the relevant PWM pin.

jens1o, 2019-02-25:
Oh, I'm so sorry that I didn't looked into the example folder. I'm feeling so terribly stupid now.

I'll use an external power supply, just in case. Thanks for the heads up!

Thank you so much!

closed 2019-02-25 by jens1o