golemparts/rppal #24: embedded hal gpio InputPin OutputPin traits

jacobrosenthal, 2019-03-09:
Was writing a bit banged proprietary serial implementation and found your lib was the only one that could perform. I was hoping to embedded hal traits which are implemented but to my eyes not for InputPin and OutputPin. Any reason not to? 

Thanks!

golemparts, 2019-03-10:
Thanks for the question! Currently RPPAL only implements the "proven" embedded-hal traits. For the GPIO module that would be `digital::OutputPin` and `PwmPin` for `OutputPin` and `IoPin`.

The remaining GPIO related traits are all `unproven` traits, unless I missed one by mistake. Initially I was hesitant to implement those, since they don't follow semver rules and could introduce breaking changes even in patch releases. However, since many of them make sense to add support for, I've decided to put them behind an additional opt-in feature flag as part of the 0.11.2 release that's currently in development. I'll be starting on those after I'm finished with the UART module, likely in the next week or so.

jacobrosenthal, 2019-03-10:
Thanks for the quick reply. Ill be watching and give a test when I see it come through

closed 2019-03-10 by golemparts

golemparts, 2019-03-11:
@jacobrosenthal I've implemented all `unproven` embedded-hal traits for `Pin`, `InputPin`, `OutputPin` and `IoPin` where applicable, on the master branch. Feel free to give them a try by enabling the `hal-unproven` feature flag.

I still need to go through the other peripherals, as well as add generic delay/rng implementations, but this should help you with your project.

jacobrosenthal, 2019-03-12:
The input pin is working for my uses
https://github.com/jacobrosenthal/rust-bitbang-wiegand

jacobrosenthal, 2019-03-12:
Though I had to drop back the libc version so I could use websockets
https://github.com/jacobrosenthal/rppal/commit/31c2a6a8a561568d1c5a6234fddd21f9db39c4b2

How is this thing generally handled in the Rust ecosystem?

golemparts, 2019-03-12:
Glad to hear the `unproven` traits are helpful.

Regarding the libc version, assuming the libc crate follows semver rules, 0.2.50 doesn't contain any breaking changes compared to 0.2.40. The "0.2.40" libc version dependency for ws-rs is equivalent to "^0.2.40", which also matches 0.2.50, so there shouldn't be a need to downgrade rppal's libc dependency. Cargo should be using 0.2.50 for both. What kind of issues/errors did you run into?

jacobrosenthal, 2019-03-12:
```
error: failed to select a version for `libc`.
    ... required by package `rppal v0.11.1 (https://github.com/golemparts/rppal.git#f25e302a)`
    ... which is depended on by `bitbang-wiegand v0.1.0 (https://github.com/jacobrosenthal/rust-bitbang-wiegand.git#0ee1fb23)`
    ... which is depended on by `littlebird-backend v0.1.0 (/Users/jacobrosenthal/Downloads/access-control-software/backend)`
versions that meet the requirements `^0.2.50` are: 0.2.50

all possible versions conflict with previously selected packages.

  previously selected package `libc v0.2.43`
    ... which is depended on by `rppal v0.11.1`
    ... which is depended on by `littlebird-backend v0.1.0 (/Users/jacobrosenthal/Downloads/access-control-software/backend)`
```

golemparts, 2019-03-12:
I'm not sure what would cause that issue in this situation, unless either littlebird-backend or one of its other dependencies has a strict libc requirement not to go passed 0.2.43. ws-rs requires ^0.2.40, so it should build fine with rppal's ^0.2.50 requirement.

I haven't run into this kind of version conflict before myself, but based on what you've shared I would look for the source of the conflict in another crate that also depends on libc besides rppal and ws-rs. I suspect you haven't run into issues with ws-rs before, because its ^0.2.40 requirement is compatible with the 0.2.43 requirement of this suspected other crate. You might be able to find the cause by searching your `Cargo.lock` file for libc dependencies. Perhaps a `cargo clean` and `cargo update` will help update any older libc dependencies if none of the other crates have a strict =0.2.43 requirement.

I just built a basic test project that depends on both the current rppal version on GitHub, and the latest ws-rs with all features enabled so libc is included. Cargo selected libc 0.2.50, and I didn't encounter any errors while building the project.