golemparts/rppal #27: Stubbing?

Aehmlo, 2019-04-10:
I'd like to be able to enable a `stub` feature (or whatever) and be able to use `rppal` GPIO functions on CI (which obviously isn't a Raspberry Pi). I [currently](https://github.com/Aehmlo/deoxy/blob/restructure/src/pin.rs) implement this myself, but it'd be great to have this capability upstream (and thereby reduce the surface area of my own code).

golemparts, 2019-04-10:
Thanks for the suggestion! Mocking the various peripherals in software has crossed my mind as well, since it would allow me to finally add somewhat meaningful tests. I've held off on working on it so far, because rppal wasn't feature-complete until recently. Now that I've finished adding support for the final peripheral (UART), adding mocking for all peripherals through a feature flag would fit nicely on the roadmap to v1.0.0. I'll add it to the feature request list as something to work on after the upcoming 0.11.2 release.

Aehmlo, 2019-04-10:
If you'd like a hand with it, I'd be happy to help out!

golemparts, 2019-04-11:
Thanks, I appreciate the offer! I'm not quite ready to dive into this right now, but I would like to consider the relevant implementation details beforehand to keep them consistent across all peripherals, especially with the large amount of changes required throughout the entire project, so I'll hold off on asking for PRs for now. I'll definitely contact you if I need a hand when this moves to active development, or to discuss the proposed implementation.

JohnRudolfLewis, 2020-02-20:
Any update on this? I would really like to write unit tests for my code that uses Gpio and other parts of rppal.

ryankurte, 2020-06-19:
Hey, it's not directly `rppal` / doesn't cover all of `rppal`, however, where you're consuming `embedded-hal` traits implemented / exposed by `rppal` you may find [dbrgn/embedded-hal-mock](https://github.com/dbrgn/embedded-hal-mock) useful.

golemparts, 2024-11-28:
Closing this to cleanup old feature requests that are unlikely to be implemented, as this is currently outside of RPPAL's scope. The embedded-hal mock crate mentioned above is a good alternative if you're primarily working with embedded-hal.

closed 2024-11-28 by golemparts