golemparts/rppal #32: Is setting alt mode necessary to use PWM?

mwillsey, 2019-06-24:
Is setting a pin to the correct alt mode necessary to use hardware PWM?

The [example](https://github.com/golemparts/rppal/blob/master/examples/pwm_blinkled.rs) doesn't seem to do that, which strikes me as confusing. It doesn't look like the source code for PWM creation sets the alt mode under the hood. Aren't all GPIO pins set to "input" mode by default?

Thanks!

golemparts, 2019-06-25:
Thanks for the question! The appropriate alt modes are automatically set by enabling pwm through `/boot/config.txt`, so there's no need to set them through RPPAL. See the [docs](https://docs.golemparts.com/rppal/0.11.3/rppal/pwm/index.html) for more information on the settings that are needed to enable hardware PWM.

mwillsey, 2019-06-25:
Thanks, that answers it perfectly

closed 2019-06-25 by mwillsey