golemparts/rppal #35: crate doesn't build on non-unix systems

xobs, 2019-08-29:
Somewhat related to #27 -- if you include this crate in a project, it won't build on any other platforms.

This can be worked around by adding a Cargo-level conditional.  However, it would be nicer to modify `lib.rs` such that it only builds any modules that are compatible with a given platform.

The `nix` crate does something similar, allowing unconditional inclusion on all platforms:

https://github.com/nix-rust/nix/blob/master/src/lib.rs

xobs, 2019-08-29:
Note that it also fails on Darwin, linux-mips, and linux-mips64: https://travis-ci.org/xobs/wishbone-utils/builds/578424583

golemparts, 2019-09-09:
Thanks for reporting this issue. Since rppal won't work on anything other than a Raspberry Pi running a supported linux distro, adding a Cargo-level conditional would be the appropriate workaround when working with larger projects that target additional platforms. However, I do like the idea of a cleaner solution where that step can be omitted. I'll look into the solution used by `nix`.