golemparts/rppal #36: Can 'Rock PI 4' be supported ？

danclive, 2019-09-09:
I found 'Rock PI 4' is also a good development board, with hardware interface compatible with raspberry pi. Can it be supported?

Rock PI 4 Home page: http://rockpi.org/
                           wiki:   https://wiki.radxa.com/Rockpi4

golemparts, 2019-09-09:
Thanks for the suggestion. Some of rppal's peripheral support relies on standardized interfaces provided by linux, which will likely be compatible with the Rock Pi 4, although they would still need to be tweaked specifically for the peripherals available on the Rock Pi. However, the GPIO module directly accesses memory locations which are unique to the Raspberry Pi. The Rock Pi 4 uses a different SoC which likely won't function in the same way. Adding support for the GPIO peripheral would require an additional GPIO module specifically developed for the Rock Pi 4 (or one that relies on slower linux drivers), which would also need to be maintained for future releases.

I'm not in a position to work on this myself since I don't own a Rock Pi 4, but if anyone else has access to the platform, and is willing to invest some time in figuring out what would be needed to add support to rppal, I would be interested in hearing what changes would be required to make it work.

danclive, 2019-09-09:
Thanks. I will try to use 'rppal' on 'Rock PI 4' later. If I find anything, I will give feedback.