golemparts/rppal #37: Can not use embedded-hal features

rumatoest, 2019-09-20:
There is a possibility that this is not a bug and I'm just a dumb coder but right now I believe in opposite. 

I've implemented DHT sensor driver based on HAL and I want to test it on RPI via rppal.
https://github.com/rust-rum/dht-hal-drv-test-rpi

But I've found that I'm not able to use HAL traits in my code. 
Looks like in your code all traits implementation for HAL are in non public module.

Majkl578, 2019-09-20:
You need to opt-in for those using feature flags: https://github.com/golemparts/rppal#usage

rumatoest, 2019-09-20:
Actually I've opted out all features I need, but still I've got an error
```
expected trait embedded_hal::digital::v1::InputPin, found struct `rppal::gpio::pin::InputPin`
```

closed 2019-09-21 by rumatoest