golemparts/rppal #40: HAL Delay implementation should not use thread::sleep

rumatoest, 2019-11-27:
Hi.

I've found out that you are using `thread::sleep` for DelayUs impl.
It would not work in cases where it is mostly nedded. Because such approach would not provide exact delays. I've tried to use `thread::sleep` several times in my serial like bus implementation and it failed. 

I suggest you to use https://crates.io/crates/spin_sleep inside delay implementations, thus it will be possible to use this API to interact with hardware.

golemparts, 2019-11-28:
Thanks for the suggestion! The current [SoftPwm](https://github.com/golemparts/rppal/blob/master/src/gpio/soft_pwm.rs) implementation uses a similar approach to spin_sleep. I'll see if I can use that for DelayUs as well.

closed 2021-08-19 by golemparts

graphex, 2021-09-04:
Ha i just implemented a `spin_sleep` `Delay` to fix this issue on my project and was coming here to see if I should make a PR! Any idea when this will be in a release?

golemparts, 2021-09-04:
@graphex I want to get a new release out the door soon because of the new embedded-hal 1.0.0-alpha support, so I expect a 0.13.0 release to be available within the next 1-2 weeks.

commit 3f6b615d (2021-08-19): Merge pull request #93 from reitermarkus/embedded-hal-1.0

Add support for embedded_hal 1.0.0-alpha.4

Closes #40