golemparts/rppal #42: PWM help

audstanley, 2020-02-10:
[image]

Just need a little help, I'm trying to have a 1kHz frequency, 10% duty cycle, and I've got cross compilation all set up, and since I'm brand new to Rust, could you point me in the right direction on how I enable the pwm "Result."  I'm still trying to figure out Results, Options, and all that... not sure what to do next.  Thanks, I'm going to get a book, but in the meantime it would be amazing for some quick help. :-)

audstanley, 2020-02-10:
I figured it out, and I'm getting an I/O error that the file doesn't exist.  I'll create the udev rule, and go from there.

golemparts, 2020-02-13:
Glad to hear you figured it out! Rust takes some getting used to, but it offers a lot of benefits once you do. You may find the PWM examples in the ```examples``` directory useful.

Are you still getting an I/O error? If you are, make sure you're running a recent Raspbian release, and PWM is enabled in your ```/boot/config.txt```. If it's enabled, check if the directory ```/sys/class/pwm/pwmchip0/``` exists, and the current user is a member of the ```gpio``` group.

audstanley, 2020-02-15:
I am no longer getting an I/O Error!! The udev rule in the docs worked perfectly, hopefully this will help someone out later if they are just hoping the library will work out of the box without a change in permissions.  I have the raspberry pi tool chain installed on my laptop, so cross compiling works like a charm, I'm using the uname crate to make sure the build machine is correct before calling the library to built ... Maybe I'll fork, and try to add this as a feature to you library if I get some time ... It makes it nice to write more complex code, and be able to test other things before ACTUALLY scp-ing the binary over to the pi. 🙂 (also adds a small slice of overhead, so maybe not what you would be looking for ). Thank you so much for building this!  Coming from a heavy Node background, I'm falling in love with Rust.

closed 2020-02-15 by audstanley