golemparts/rppal #49: input debounce support

TheBerkin, 2020-06-03:
I'm using this library on a Pi Zero W to interface with an old rotary phone, and switch transitions definitely aren't clean 100% of the time.

Previously I have used the `gpiozero` Python library, which has the option to "debounce" inputs: a user-supplied time delay controls how soon before new input states can be registered. As far as I can tell, this crate does not have any such capability.

What I would love to see is a new method for `InputPin` that sets a debounce time using a `Duration`. This might work something like the following:
```rust
const BCM_INPUT: u8 = 2;
let gpio = Gpio::new()?;
// Ignore state changes on `button` less than 50 ms apart.
let debounce_time= Duration::from_millis(50);
let button = gpio.get(BCM_INPUT)?.into_input_pulldown().debounce(debounce_time);
// ...
```

Of course, I could implement my own debouncing code on top of `rppal`, but I strongly feel that this is something that should be implemented in this crate directly, as debouncing digital inputs from switches is an extremely common practice.

ShayBox, 2020-06-12:
This would be a good addition, adding a resistor in line does this too

golemparts, 2024-08-08:
Debounce has been added to synchronous and asynchronous interrupts for the upcoming 0.19.0 release.

closed 2024-08-08 by golemparts