golemparts/rppal #51: Internal pull resistors do not always work on Raspberry Pi 4B

TheBerkin, 2020-06-29:
It appears that attempting to activate internal pull resistors on certain pins does not work on the Raspberry Pi 4B. Specifically, I have noticed that internal pull-up resistors fail to activate when using `Pin::into_input_pullup()` on BCM pins 17, 27, and 22. The pins remain floating, resulting in undefined behavior on the inputs.

I tested the pins in Python using `gpiozero` and confirmed that they behave correctly there when using internal pull-up resistors, so the problem seems to be isolated to rppal.

I did some digging and found [this C library](http://abyz.me.uk/rpi/pigpio/examples.html#Misc_minimal_gpio) which caught my attention. On line 117 you'll find the function used to set the internal pull resistors. It appears to have a branch with separate logic for the BCM 2711 SoC (used in the Pi 4B) that appears to modify the GPIO memory in a completely different way.

Perhaps this is an incompatibility problem?

TheBerkin, 2020-06-30:
Looks like this problem does not happen for me on a Model 3B, so now I'm positive this is a compatibility problem.

Dragonrun1, 2020-09-25:
pin 18 is also not working so would say pull-ups on RPi4 just aren't working correctly in general.

PR #57 (2020-09-27): Fix pull up down for rpi4
Fixes #51 
I've tested this with a Raspberry Pi 4 and it work. Code for other models doesn't have any real changes but was not tested. This includes changes from my ioctl warning PR as well.
commit f2c60e6d (2020-09-27): Merge pull request #57 from Dragonrun1/fix_pull_up_down_for_rpi4

Fix pull up down for rpi4
Fixes #51
Closes #56
closed 2020-09-27 by golemparts