golemparts/rppal #58: Uart set custom speed of 100kbs or 100000

alex5250, 2020-10-18:
Using the S Bus protocol it required to use speed of 10000bps. Quick look in lib code make sure that all speeds are settings by macro. It will great to opportunity to add custom speeds in lib. Is it possible? 

```Rust 

 Uart::with_path("/dev/ttyAMA0", 100000, Parity::Even, 8, 2).expect("COULD SET UART");
```
Gives me error 
```Rust
thread 'main' panicked at 'COULD SET UART: InvalidValue', src/main.rs:8:69
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

golemparts, 2021-02-14:
Thanks for the suggestion. Unfortunately this is a limitation of the underlying ```termios``` functions ```cfsetispeed()``` and ```cfsetospeed()``` which are used to set the speed, and only support the currently available values. ```termios2``` does support custom speeds, but isn't officially supported for ```musl``` targets, which would cause incompatibility issues depending on which ```libc``` is used. However, since things may have changed since I last worked on the uart code, I'll take another look after 0.12.0 has been released.

alex5250, 2021-02-16:
Maybee [this ](https://gist.github.com/kennethryerson/f7d1abcf2633b7c03cf0) snippet of code will be helpful.

closed 2025-07-01 by golemparts