golemparts/rppal #63: Raspberry PI as i2c follower?

nordseele, 2020-11-20:
Hi, 
Using Rppal, is it possible to make the Raspberry Pi (3B or Zero) act as a i2c follower too ? 

I know that kind of behavior is possible with the Pigipio library and its [Bsci2c](http://abyz.me.uk/rpi/pigpio/python.html#bsc_i2c ) function but now that I switched to Rust I'm wondering if something similar with Rppal.

Thank you.

golemparts, 2021-02-14:
Thanks for the suggestion! Theoretically if pgpio can do it, so could rppal. However, one of the main differences with rppal is that the pigpiod daemon the client communicates with runs with superuser privileges, which is something I've been actively trying to avoid, but might be the only way to accomplish this. I'll add it to the list of feature requests and will look into it more once 0.12.0 has been released.

Considering this issue was opened several months ago, you may have already figured out how to do this in Rust. If so, I'd love to hear it!

nordseele, 2021-02-14:
Thanks for your reply. No, I didn’t really look any further how to do that with Rust, I asked on the official forum but didn't get any answer. In the meantime, I split my program in two, on one hand the program coded in Rust and on the other a C++ program running alongside and using the Pigpio library to handle the "i2c follower" side.

But I still use Rppal for sending data on the i2c bus and it would be great to be able to group everything in the Rust  program. Glad to read that you're considering this feature for a future version of Rppal. I hope it's feasible. Thank you very much.

closed 2025-07-01 by golemparts