golemparts/rppal #71: Can't create Gpio on Raspberry Pi 4 8GB with 2020-08-20-raspios-buster-arm64

d3249, 2021-01-31:
Working with [this OS](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/) I can't create `Gpio::new()?;`, getting this error

```rust
Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })
```
I've tracked the file to src/gpio/ioctl.rs:38

On source code there is

```rust
const PATH_GPIOCHIP: &str = "/dev/gpiochip";
```

but on my system there are these

```bash
$ ls /dev/gpiochip*
/dev/gpiochip0  /dev/gpiochip1
```

I'm using a raspberry pi 4 8GB

```bash
$ uname -a
Linux pocketvac 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
```

haku956, 2021-02-06:
Same issue here using 4B 4GB with 5.10.13

orfest, 2021-02-07:
+1 to the issue

haku956, 2021-02-14:
If this is blocking you, as it is for me, one could temporary use: `rppal = { git = "https://github.com/ryanfuji/rppal.git", branch = "patch-1" }` in your `Cargo.toml`. This points to #70.

golemparts, 2021-02-14:
The current master branch has been updated to recognize all currently available Raspberry Pi models. You should be able to use that in your projects until the next release. Let me know if you run into any additional problems.

It has been a while since I've had time to work on this project, and I need some time to sort out any remaining required changes, and update the documentation. 0.12.0 should be good to go soon though.

closed 2021-02-14 by golemparts