golemparts/rppal #73: Error: Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })

RNGKing, 2021-02-17:
When I run the 0.11.3 build on a Raspberry Pi I get a weird error.

Cargo.toml
```
[package]
name = "gpio_test"
version = "0.1.0"
authors = ["pi"]
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
rppal = "0.11.2"
```

Running the example code yields the following results:
```
use std::error::Error;
use std::thread;
use std::time::Duration;

use rppal::gpio::Gpio;
use rppal::system::DeviceInfo;

// Gpio uses BCM pin numbering. BCM GPIO 23 is tied to physical pin 16.
const GPIO_LED: u8 = 23;

fn main() -> Result<(), Box<dyn Error>> {
    println!("Blinking an LED on a {}.", DeviceInfo::new()?.model());

    let mut pin = Gpio::new()?.get(GPIO_LED)?.into_output();

    // Blink the LED by setting the pin's logic level high for 500 ms.
    pin.set_high();
    thread::sleep(Duration::from_millis(500));
    pin.set_low();

    Ok(())
}
```

The results are in the attached image.

[image]

Whenever I test the GPIO with the pre-loaded python library I have full access to the pins. Is there something I'm missing or is there some permission that I need to give?

RNGKing, 2021-02-17:
EDIT : Raspberry Pi 4B

RNGKing, 2021-02-17:
Second Edit : my provided Cargo.toml file reads incorrectly, it doesn't work on that version either.

golemparts, 2021-02-17:
This is fixed in the current master branch, and will be part of the ``0.12.0`` release, which will be available soon.

For a temporary fix, you can point your rppal dependency to the github repo until the next version is released by replacing the line in your ``Cargo.toml`` with ``rppal = { git = "https://github.com/golemparts/rppal.git" }``

RNGKing, 2021-02-17:
@golemparts Thanks for the response. I was pulling my hair out thinking that I might have broken something on my end of things.

closed 2021-02-17 by RNGKing

bammi, 2021-02-25:
Thank you @RNGKing that fixed the problem for me too for Raspberry PI 4 #75 Appreciate your help,

cheers, bammi

drauschenbach, 2021-08-07:
I am still experiencing this error message in `0.12.0`, using a Pi 4 8GB and latest Raspberry OS 32-bit. Although I'm using the Pwm functions, not the Gpio functions.

> Error: Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })

Code:

```rust
let pwm = Pwm::with_frequency(Channel::Pwm0, 25000_f64, 1_f64, Polarity::Normal, true)?;
```

drauschenbach, 2021-08-07:
I was missing a `/sys/class/pwm/` directory. I found how to enable the missing driver [here](https://www.raspberrypi.org/forums/viewtopic.php?t=287786).

```sh
$ vi /boot/config.txt
dtoverlay=pwm,pin=18,func=2
$ sudo reboot now
```