golemparts/rppal #75: Exceptions on Blink example on Raspberry PI 4

bammi, 2021-02-20:
Raspberry PI 4 GB full up to date
Linux pc-raspi41 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

Simple blink example exceptions (see attached main.rs.txt)
Traceback:

pi@pc-raspi41:~/src/rust/rppal_blink $ cargo run
   Compiling rppal_blink v0.1.0 (/home/pi/src/rust/rppal_blink)
    Finished dev [unoptimized + debuginfo] target(s) in 1.76s
     Running `target/debug/rppal_blink`
Blinking an LED on a Raspberry Pi 4 B.
Error: Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })

Tried with all: straight, hal, hal-unproven in dependencies
[dependencies]
#rppal = "0.11.3"
rppal = { version = "0.11.3", features = ["hal-unproven"] }

running with sudo makes no difference

New to Rust, any advise much appreciated

cheers,
bammi

[main.rs.txt](https://github.com/golemparts/rppal/files/6015802/main.rs.txt)

RNGKing, 2021-02-25:
Link against the github link instead of the direct version callout. 

[Here is a link to a reply by the maintainer.](https://github.com/golemparts/rppal/issues/73#issuecomment-780329963)

bammi, 2021-02-25:
Thank you RNGKing that fixed the problem: in Cargo.toml dependency:
           rppal = { git = "https://github.com/golemparts/rppal.git" }

cheers,
bammi

closed 2021-02-25 by bammi