golemparts/rppal #78: Trouble with Pi400

Papounet94, 2021-03-24:
Hello, 

I am trying to use 'rppal' crate on my Pi400 and connected it to a breadboard to perform simple tests. 
I first tried the simple examples in the crate doc to lit an LED, but got an "Io error ; No such file ...}
Then I tried the **status.rs** example from the examples folder. I compiles fine but upon running I get the following error:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/gpiotest`
Error: UnknownModel
The terminal process terminated with exit code: 1

```
My Rust configuration is up to date as is the Pi 400 one.
When I try to use the breadboard with a simple python program using Gpiozero, everything goes fine.
What is wrong in my configuration for using PI400 GPIOs with Rust?

Thanks in advance.
C. Brunel

Papounet94, 2021-03-24:
Additional comment:
I tried the **blinkled.rs** example and I got the same error as with my previous simple attempts
```
   Compiling gpiotest v0.1.0 (/home/pi/DEV/Rust/projects/gpiotest)
    Finished dev [unoptimized + debuginfo] target(s) in 2.12s
     Running `target/debug/gpiotest`
Error: Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })
```

bammi, 2021-03-24:
I had reported the same issue with an RPI 4, which is essentially what the 400 is. Turns out the fix is to link to the current version of rppal

See my closed issue here:
https://github.com/golemparts/rppal/issues/75

and the direct  solution link here
https://github.com/golemparts/rppal/issues/73#issuecomment-780329963

after adjusting the dependency in Cargo.toml the blink example worked perfectly

if you are using the rust gpiozero, there is similarly a edge branch of rust_gpiozero that also in turn points to the correct version of rppal, otherwise you get the failure you are seeing on any model of RPI 4/400/Compute Module 4 etc. To us that version of of rust_gpiozero, please point to
[dependencies]
rust_gpiozero = { git = "https://github.com/rahul-thakoor/rust_gpiozero", branch = "rppal-edge" }

hope that helps,
cheers,
bammi

Papounet94, 2021-03-24:
Hi Bammi, 

Your fix worked well for most of the examples I have tried (blink, blink_signal, status). Not sure that I fully understands the reason why, but whatever ...
However the PWM blink example failed with the same IO error as previously.
```
Error: Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })
```
I will try to dig down a bit into why I cannot access the PWM pins.

Did not tried the rust_gpiozero crate yet.

Thanks a lot for your help.
Regards.
Christian

Papounet94, 2021-03-24:
OK. I have found out why I cannot access PWM channels. 
It seems that I need to enable the PWM controller at boot time, by tampering with **/boot/config.txt** file.
Will try it later on.

Thank again for your time
Christian.

golemparts, 2021-04-14:
Closing this as this is resolved in the master branch. 0.12.0 should be released soon.

closed 2021-04-14 by golemparts