golemparts/rppal #83: Mismatched Types on Master Branch

TheCatster, 2021-04-13:
Hi all,
As with the latest issues with the Pi 4 that require using the master branch, I have switched over accordingly (I believe that was clarified in issue #73 or #75.) However, when I attempt to build any project with `rppal` as of the last week, cargo exits with an error.

Steps to reproduce:
- Require `rppal` from git (specifically, I did `rppal = { git = "https://github.com/golemparts/rppal.git" }`)
- Attempt to run/build/check with cargo (essentially try to compile)
- Errors with this message:
```rust
error[E0308]: mismatched types
  --> /home/danya/.local/share/cargo/git/checkouts/rppal-451d1b4d0438fc03/a1ab827/src/pwm/sysfs.rs:59:17
   |
59 |                 buf.as_mut_ptr(),
   |                 ^^^^^^^^^^^^^^^^ expected `i8`, found `u8`
   |
   = note: expected raw pointer `*mut i8`
              found raw pointer `*mut u8`
```

This has occurred on all projects, from simple to complex, across many computers, so I don't believe it is simply an issue of my setup. This has popped up within the past week, but I hadn't seen any commits since February. I know this issue was to have been fixed in version `0.12.0`, any update on that?

I'll provide whatever else I can if needed.

golemparts, 2021-04-14:
Thanks for your report. I haven't been able to reproduce the same error. However, I've changed the array to use libc's ```c_char``` which should result in using the correct type in all circumstances. Let me know if you still run into this issue after the change has been committed.

Unfortunately I haven't had a chance to work on this project in a while. I wanted to clean up the documentation before releasing the new version, but considering the latest release doesn't work on the Pi 4 I'll schedule a release for 0.12.0 and save any low priority updates for 0.12.1.

closed 2021-04-14 by golemparts

TheCatster, 2021-04-14:
Strangely enough, I am still getting the same error. I have done `cargo update` to pull down the latest commit, and it still results in an identical error message. I went to go build this on another machine, and it still happened. Note I've also pinned the dependency on your commit where you fixed the issue, and still no change.

TheCatster, 2021-04-14:
Interesting, I am wrong. The error *has* changed. Now, the issue is only in the second usage (see line 91). If I do the same cast, then it works with no issue. The previous issue with line 57 has been resolved.

commit 4d06efc3 (2021-04-14): Use libc::c_char for buffer instead of u8

Fixes #83
PR #84 (2021-04-14): Cast buf as c_char
You did this the first time, but forgot on the second. This fixes all PWM errors for me. Closes #83 completely this time.