golemparts/rppal #87: Link pin number website in the example

williamdes, 2021-04-17:
Hi !
You example works great, that said it would need a reference to somwhere whre to know what pin number is what

https://fr.pinout.xyz/pinout/pin4_alimentation_5v

williamdes, 2021-04-17:
That said only the French website talks about `BCM`

golemparts, 2021-04-17:
Thanks for the suggestion! The numbering system used by rppal is the internal numbering of the GPIO lines as set by Broadcom, which is unrelated to their physical position on the Raspberry Pi's GPIO header. They are referred to as 'BCM' to make it clear these aren't physical locations, or simply as 'GPIO' like on pinout.xyz.

I think it would be a good idea to add a bit more information for people that aren't familiar with this numbering scheme. Adding details of the GPIO header for every Raspberry Pi doesn't fit within the scope of this project, but adding a reference to an external site like pinout.xyz, or a diagram of the GPIO header for the most recent Raspberry Pi model, should help a lot.

williamdes, 2021-04-17:
Off topic (but related): Oh snap just found out about: https://raspberrypi.stackexchange.com/a/106904/114760

golemparts, 2021-04-17:
Yeah, you're definitely limited in what you can use those for. I mainly use the GPIO pins to send control signals to something that's powered by the 3.3V or 5V power pins. Are you building anything in particular, or just learning how to use a Pi?

williamdes, 2021-04-17:
> Yeah, you're definitely limited in what you can use those for. I mainly use the GPIO pins to send control signals to something that's powered by the 3.3V or 5V power pins. Are you building anything in particular, or just learning how to use a Pi?

I quickly built https://github.com/wdes/temprpi to power a fan that was delivered in the case, but just when I want and not all the time. For now it does not work but the case manufacturer says that the fan can run on a 3.3v pin but it does not work. Strangely when I run the binary the fan like moves 0.5sec so I will have to continue figuring out what is wrong

(just trying to not install Python ^^)

But in the past I had this problem if I recall correctly and had bought a relay board but never used it. And for now the board is not with me.

Conclusion: I will try to figure out if I did something wrong with my Rust code or not.

golemparts, 2021-04-17:
Good luck on the project! I can't offer you any specific advice, cause every fan is different. I would be careful with powering a fan directly from a GPIO fan though. As mentioned in that post you linked, they can't handle much. You can probably find circuits other people have used with 3.3V fans. Once you have things up and running with a regular GPIO pin, you could have some fun with PWM to control the fan speed. 😄

williamdes, 2021-04-17:
```sh
docker run --rm --device /dev/ttyAMA0:/dev/ttyAMA0 --device /dev/mem:/dev/mem --privileged -ti python:2 /bin/sh
apt-get update
apt-get install git-core sudo
git clone https://github.com/WiringPi/WiringPi
cd WiringPi/
./build
gpio readall
```

I really do not understand, the 3.3v pin works but none of the other pins work even if I put them in OUT mode and up :/

```
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | OUT  | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | OUT  | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | OUT  | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
```

> Once you have things up and running with a regular GPIO pin, you could have some fun with PWM to control the fan speed. smile

Yeah, sure 😄 

> Good luck on the project! I can't offer you any specific advice, cause every fan is different.

Sure ^^ 
Will continue to research and use your lib

golemparts, 2021-04-17:
> I really do not understand, the 3.3v pin works but none of the other pins work even if I put them in OUT mode and up :/

First thing I would do is verify the GPIO pin is outputting by measuring with a multimeter, or if you don't have one available, an LED and a resistor. If the GPIO pin is working fine, perhaps try a setup with an NPN transistor where the fan is powered by the 3.3V pin, and the GPIO pin turns the transistor on and off. If you google for `raspberry pi control fan` you should find some circuit examples.

williamdes, 2021-04-17:
> First thing I would do is verify the GPIO pin is outputting by measuring with a multimeter, or if you don't have one available, an LED and a resistor. If the GPIO pin is working fine, perhaps try a [...]

I now understand after reading https://www.raspberrypi.org/forums/viewtopic.php?t=178461 why my fan was like moving, starting, moving, starting when the pin was UP.
Thank you for helping me with that, I will wait to have my [relay board](https://raspberrypi-tutorials.fr/raspberry-pi-commutateur-de-relais-de-commande-via-gpio/) and that will do the scenario you described :)

williamdes, 2021-08-30:
Currently running a PWM Noctua fan, closing this one :)

closed 2021-08-30 by williamdes