golemparts/rppal #88: Set initial logic level value before changing a pin mode to output

golemparts, 2021-04-23:
Either change ``into_output```, or add an additional method that sets a pin's logic level before changing the mode. This will help in cases where an attached device behaves in an undesirable way if the pin starts out as low.

kalj, 2021-04-23:
I think it would make most sense to have something like a `into_output_with_level`. Seems cleaner than adding a setter to `Pin`.

golemparts, 2021-04-23:
Agreed. I'm thinking of either adding ```into_output_with_level```, or ```into_output_low``` and ```into_output_high```. The latter would be more consistent with how input pins have ```into_input_pullup``` and ```into_input_pulldown```. If I went with the ```_with_level``` version, I'd feel the need to also change the pullup/down methods, which would be a breaking change.

kalj, 2021-04-23:
Sounds great with `into_output_low` + `into_output_high` and being consistent with input pull up/down settings.

closed 2021-05-10 by golemparts

kalj, 2021-05-14:
Great! Thanks a lot for adding this!

commit 95462f05 (2021-05-10): Add Pin::into_output_low() and into_output_high()

Closes #88