golemparts/rppal #90: set_mode does not change mode of input pin

BrianHoldsworth, 2021-06-01:
Hi. I am using rppal on Linux 5.10 with a RPi Zero W. I'm trying to enable serial hardware handshaking using RTS/CTS mode of BCM GPIO pin 16/17. The pins must be set to Mode::Alt3.

```
	// Flow control initially disabled
	let mut cts0 = Gpio::new()?.get(GPIO_CTS0)?.into_io(Mode::Input);
	let mut rts0 = Gpio::new()?.get(GPIO_RTS0)?.into_io(Mode::Input);
	// Enable CTSRTS flow control for serial port
	cts0.set_mode(Mode::Alt3);
	rts0.set_mode(Mode::Alt3);
```
But the calls to `set_mode` seem to have no effect, and hardware handshaking remains disabled.

BrianHoldsworth, 2021-06-01:
Er, disregard. I figured out that dropping a Pin returns it to the default mode automatically. That's probably the way it should work.

closed 2021-06-01 by BrianHoldsworth

golemparts, 2021-06-02:
It does indeed reset its mode by default. However, if you'd prefer to keep the mode set after dropping the pin, you can use ```set_reset_on_drop(false)```. See https://docs.golemparts.com/rppal/0.12.0/rppal/gpio/struct.IoPin.html#method.set_reset_on_drop