golemparts/rppal #91: Concurrency causes DoS of interrupts

TheAlgorythm, 2021-07-24:
I am using rppal v0.12.0 with an Raspberry Pi 4B and found a situation where the library isn't responding to interrupts.
I'm pretty sure that it is a concurrency bug, like a `Mutex` lock.
The following minimal reproducible example responds only alternating between the pins and doesn't continue on every falling edge.
```rust
use rppal::gpio::{Gpio, Trigger};
use std::thread;
use std::time::Duration;

fn event_loop(gpio: &Gpio, pin: u8) -> thread::JoinHandle<()> {
    let mut dev = gpio.get(pin).unwrap().into_input_pulldown();
    thread::spawn(move || {
        println!("init {} done", pin);
        let mut counter = 0_usize;
        loop {
            println!("await {} off {}", pin, counter);
            dev.set_interrupt(Trigger::FallingEdge).unwrap();
            dev.poll_interrupt(true, None).unwrap();
            println!("awaited {} off {}", pin, counter);
            thread::sleep(Duration::from_millis(100));
            counter += 1;
        }
    })
}

fn main() {
    let gpio = Gpio::new().unwrap();
    let first_el = event_loop(&gpio, 6);
    event_loop(&gpio, 16).join().unwrap();
    first_el.join().unwrap();
}
```

golemparts, 2021-07-24:
Thanks for reporting this issue. This is, unfortunately, by design. As mentioned in the documentation:

> Calling ```poll_interrupt``` blocks any other calls to ```poll_interrupt``` (including on other ```InputPin```s) or ```Gpio::poll_interrupts``` until it returns. If you need to poll multiple pins simultaneously, use ```Gpio::poll_interrupts``` to block while waiting for any of the interrupts to trigger, or switch to using asynchronous interrupts with ```InputPin::set_async_interrupt```.

```poll_interrupt``` was designed as the easiest way to block while waiting for an interrupt to trigger on a single pin. The other options were added as alternatives when multiple pins need to be polled, either synchronously (```Gpio::poll_interrupts```) or asynchronously (```set_async_interrupt```).

I'm planning to dive back into the interrupt code eventually since I'd like to add a trigger timestamp as suggested in #89, which might be a good time to revisit this code and see if there's a better way of doing things. For now, your best bet would be the two options mentioned above.

TheAlgorythm, 2021-07-24:
Sorry that I didn't read that part of the documentation. I come from rust_gpiozero and there it isn't mentioned. As your crate supports async interupts which solve the problem, I will use rppal. I think it makes even my code cleaner.

closed 2021-07-24 by TheAlgorythm

golemparts, 2021-07-24:
No problem! Let me know if you run into any other issues.