golemparts/rppal #92: Error: Io(Os { code: 2, kind: NotFound, message: "No such file or directory" })

EdwardDeaver, 2021-07-25:
Hello, 

I am running DietPi on a Raspberry Pi 2. 

When I try to run the led pwm example I get the title error. 

I tried using "rppal = { git = "https://github.com/golemparts/rppal" }" as the dependency but it doesn't seem to work.

golemparts, 2021-07-25:
Thanks for opening this issue. A NotFound error in the pwm module usually means support for PWM hasn't been enabled yet on your Raspberry Pi. This [documentation page](https://docs.golemparts.com/rppal/0.12.0/rppal/pwm/index.html#pwm-channels) specifies how to enable the PWM channels in Raspberry Pi OS. It also shows you how to use PWM without needing to run your code with superuser privileges (```sudo```).

I haven't used DietPi myself, so I'm not sure if any additional steps are required for your distribution.

Please let me know if you're still running into the same issue after enabling the relevant PWM channel(s).

EdwardDeaver, 2021-07-27:
Thank you! That fixed it!

closed 2021-07-27 by EdwardDeaver