golemparts/rppal #107: ioctl compile errors for all arm musl targets

austenadler, 2022-07-26:
Whenever compiling for any arm musl target, I get tons of ioctl errors:

```
   Compiling rppal v0.7.1
error[E0061]: this function takes at least 2 arguments but 3 arguments were supplied
   --> /home/stonewareslord/.cargo/registry/src/github.com-1ecc6299db9ec823/rppal-0.7.1/src/i2c/ioctl.rs:328:18
    |
328 |     parse_retval(ioctl(fd, REQ_SMBUS, &mut request))?;
    |                  ^^^^^     ---------  ------------ argument unexpected
    |                            |
    |                            expected `i32`, found `u64`
    |
note: function defined here
   --> /home/stonewareslord/.cargo/registry/src/github.com-1ecc6299db9ec823/libc-0.2.126/src/unix/linux_like/linux/musl/mod.rs:725:12
    |
725 |     pub fn ioctl(fd: ::c_int, request: ::c_int, ...) -> ::c_int;
    |            ^^^^^
help: you can convert a `u64` to an `i32` and panic if the converted value doesn't fit
    |
328 |     parse_retval(ioctl(fd, REQ_SMBUS.try_into().unwrap(), &mut request))?;
    |                                     ++++++++++++++++++++
help: remove the extra argument
    |
328 |     parse_retval(ioctl(fd, {i32}))?;
    |                  ~~~~~~~~~~~~~~~~
```

## Testing method

1. Make a new bin `cargo new t`
2. Add `rppal = "0.17.1"` to `Cargo.toml`
3. Run:
```sh
for t in aarch64-unknown-linux-musl aarch64-unknown-linux-gnu arm-unknown-linux-gnueabi arm-unknown-linux-gnueabihf arm-unknown-linux-musleabi arm-unknown-linux-musleabihf armv7-unknown-linux-gnueabi armv7-unknown-linux-gnueabihf armv7-unknown-linux-musleabi armv7-unknown-linux-musleabihf; do
    echo "TARGET: ${t}"
    cargo build --target "${t}" --keep-going -Z unstable-options
done
```

## Expected

Either everything compiles, or gives linker errors due to no `.cargo/config`

## Actual

Every musl target listed fails with the above error 19 times, and every non-musl one build successfully:

- aarch64-unknown-linux-musl -> Fails
- arm-unknown-linux-musleabi -> Fails
- arm-unknown-linux-musleabihf -> Fails
- armv7-unknown-linux-musleabi -> Fails
- armv7-unknown-linux-musleabihf -> Fails
- aarch64-unknown-linux-gnu -> Succeeds
- arm-unknown-linux-gnueabi -> Succeeds
- arm-unknown-linux-gnueabihf -> Succeeds
- armv7-unknown-linux-gnueabi -> Succeeds
- armv7-unknown-linux-gnueabihf -> Succeeds

## Note

- I thought this could be an issue with the libc crate, but if you remove the rppal dependency and only add `libc = "0.2.126"`, everything builds properly
- This might be related: https://github.com/rust-lang/libc/issues/1036

golemparts, 2022-11-07:
Apologies for the late response. I'm not seeing any issues building for ```armv7-unknown-linux-musleabihf``` or ```aarch64-unknown-linux-musl``` on the latest release of the Raspberry Pi OS distro (bullseye). However, your compiler log mentions ```rppal v0.7.1```. Have you tried the current release (v0.13.1)?

If you're still getting these errors with the latest release of RPPAL and the most recent rust compiler, please provide any details on your build environment / Linux distro / hardware that may be relevant as I'm unable to reproduce those errors on a stock Raspberry Pi OS build.

golemparts, 2022-11-09:
Closing this for now. If you're still experiencing these issues, please provide any additional relevant info you can think of to help reproduce these compilation errors and we can re-open this issue.

closed 2022-11-09 by golemparts

austenadler, 2022-11-15:
Thank you. I can confirm it does build on all of those targets now. 

Looks like rppal `0.11.0` and up build fine, and `0.10.0` and below don't compile for at least one of those targets.