golemparts/rppal #120: cargo check yields errors after adding to Cargo.toml

reeslabree, 2022-12-05:
Wondering if anyone can shed some light on what's causing these errors - when I add `rppal = "0.14.1"` under the `[dependencies]` section in my `Cargo.toml` and run `cargo check` I get a whole slew of errors. Am I missing some dependencies, or did I install something wrong?

Examples:
```
error[E0432]: unresolved imports `libc::epoll_event`, `libc::EPOLLERR`, `libc::EPOLLET`, `libc::EPOLLIN`, `libc::EPOLLONESHOT`, `libc::EPOLLOUT`, `libc::EPOLLPRI`
  --> /Users/me/.cargo/registry/src/github.com-1ecc6299db9ec823/rppal-0.14.1/src/gpio/epoll.rs:11:16
   |
11 | pub use libc::{epoll_event, EPOLLERR, EPOLLET, EPOLLIN, EPOLLONESHOT, EPOLLOUT, EPOLLPRI};
   |                ^^^^^^^^^^^  ^^^^^^^^  ^^^^^^^  ^^^^^^^  ^^^^^^^^^^^^  ^^^^^^^^  ^^^^^^^^ no `EPOLLPRI` in the root
   |                |            |         |        |        |             |
   |                |            |         |        |        |             no `EPOLLOUT` in the root
   |                |            |         |        |        no `EPOLLONESHOT` in the root
   |                |            |         |        no `EPOLLIN` in the root
   |                |            |         no `EPOLLET` in the root
   |                |            no `EPOLLERR` in the root
   |                no `epoll_event` in the root
```
```
error[E0412]: cannot find type `IoctlLong` in this scope
  --> /Users/me/.cargo/registry/src/github.com-1ecc6299db9ec823/rppal-0.14.1/src/gpio/ioctl.rs:39:23
   |
39 | const SIZE_CHIP_INFO: IoctlLong = (mem::size_of::<ChipInfo>() as IoctlLong) << SIZESHIFT;
   |                       ^^^^^^^^^ not found in this scope
```
There's probably a hundred errors like this, I can list more if necessary but just wanted to provide some examples.

golemparts, 2022-12-06:
Hey @reeslabree. It looks like you're not working directly on a Raspberry Pi, in which case you'll want to configure your project for cross-compilation. An easy way to cross-compile for the Pi is to use [cross](https://github.com/cross-rs/cross). The [Cross compilation](https://github.com/golemparts/rppal#cross-compilation) section in the README also offers some additional information on how to configure your build target to make `cargo check` work.

I hope that helps!

reeslabree, 2022-12-08:
Cross rules! Anyone discovering this thread in the future, you'll take years off your life and lose 20% of your hair trying to do it the old fashioned way. Just use Cross!

closed 2022-12-08 by reeslabree