golemparts/rppal #124: is there any clear or clean or cleanup method?

krishnaTORQUE, 2023-02-10:
is there any clear or clean or cleanup method or clear the gpio pins?

golemparts, 2023-02-11:
By default, pins are reset to their original state when they go out of scope (see `set_reset_on_drop` on the various pin structs). To programmatically reset a pin to its original state, you'll have to drop any references to that pin.

Note that `drop` methods aren't called when a process is abnormally terminated (for instance when a `SIGINT` signal isn't caught), so exiting an application by pressing Ctrl-C won't reset the pin states. Check out this [example](https://github.com/golemparts/rppal/blob/master/examples/gpio_blinkled_signals.rs) for a workaround.

LeCyberDucky, 2023-08-20:
> By default, pins are reset to their original state when they go out of scope (see `set_reset_on_drop` on the various pin structs). To programmatically reset a pin to its original state, you'll have to drop any references to that pin.

How does this work for pull up/down resistors? 

I have modified my `/boot/config.txt`, such that GPIO 6 on my pi is configured as an input pin with a pull-up resistor upon boot. My multimeter tells me that this works as expected (I measure about 3.3V between this pin and GND). 

Now, I have created a program using `rppal` that reconfigures this pin to do some work. I would expect the things to return to the original state after the program has finished, but it leaves my pin at ~0.8V. The program does not crash; it finishes successfully.

golemparts, 2023-09-28:
Apologies for the late response.

> Edit 2: Welp, seems like it's simply not possible to read the state of the resistors on my 3b+ :/

Yeah, unfortunately that was the case at the time the cleanup code was added. I wasn't aware that this changed for the BCM2711 (and presumably the upcoming BCM2712) so it's something I may need to revisit.

As for your specific situation, your best bet would probably be to disable `reset_on_drop` and do any needed clean-up yourself to keep the pull-up resistor enabled.

closed 2023-09-28 by golemparts