golemparts/rppal #135: Release v0.15.1

DASPRiD, 2023-12-10:
The last release was in October, but shortly after in November, a critical fix was implemented to make Raspberry Pi Zero W 2 work again:

https://github.com/golemparts/rppal/commit/f84c36ad3b03e741323e3258bd82173c1fc5d2f3

At the moment, rrpal is broken on the Zero W 2, so a hotfix release would be appreciated.

golemparts, 2023-12-10:
Hi @DASPRiD. rppal should work fine on the Zero W 2 running Raspberry Pi OS, which is the only distro we officially support. The commit you linked contains a fix specifically for Arch Linux, which is why it wasn't deemed a critical update. Additionally, anyone that needs the fix before a release is published can make their rppal dependency use the github repo until a new version is available. If you are running Raspberry Pi OS and are currently having issues with the Zero W 2 using rppal 0.15.0, let me know, as that would be a new issue.

Considering we also have support for embedded-hal v1.0.0-rc.2 ready to go, I'll schedule a 0.16.0 release for the coming week, which will include the fix you mentioned.

DASPRiD, 2023-12-10:
There was a kernel update recently in Raspian which removed the `Hardware` key from `/proc/cpuinfo`, which is why the other detection methods are now required.

golemparts, 2023-12-10:
Thanks for pointing that out. I wasn't aware `Hardware` was removed. That does make this a critical update for the Zero W 2 at least. I'll have to address that change as well.

golemparts, 2023-12-10:
I just read the discussion on why `Hardware` was removed. Can't say I agree, but what's done is done. I pushed a fix that removes the need for this field to be present. Since this could potentially affect a lot of people, I'll try to get 0.16.0 out the door today. Thanks for letting me know about this issue.

closed 2023-12-10 by golemparts

DASPRiD, 2023-12-10:
Thanks for your quick response on that! :)