golemparts/rppal #141: No PWM on a raspberry pi 5

whattheflukz, 2024-04-04:
using the following code i am unable to produce a pwm signal on a raspberry pi 5. See attached scope probe image

Its possible i'm just doing something wrong but this code compiles and runs just fine, and is basically pulled straight from the examples

`rs
const PERIOD_MS: u64 = 10;
const PULSE_MIN_US: u64 = 500;
const PULSE_NEUTRAL_US: u64 = 1250;
const PULSE_MAX_US: u64 = 2500;
`
 `tokio::spawn(async move {
        let mut pwm = Pwm::with_period(
            Channel::Pwm0,
            Duration::from_millis(PERIOD_MS),
            Duration::from_micros(PULSE_MAX_US),
            Polarity::Normal,
            true,
        )?;
        loop {
            for pulse in (PULSE_MIN_US..=PULSE_NEUTRAL_US).step_by(10) {
                let pwm_res = pwm.set_pulse_width(Duration::from_micros(pulse));
                println!("Res: {:?}", pwm_res);
                sleep(Duration::from_millis(20)).await;
            }
        }
        Ok::<(), rppal::pwm::Error>(())
    });`
[image]

Thanks!
Keith Sachs

golemparts, 2024-04-05:
Hey @whattheflukz. Unfortunately I don't have a Pi 5 available for testing, so I can't run your code myself. I do have a few suggestions you could try to pinpoint the cause of the issue though.

Besides running a minimal synchronous version of the code and seeing if that works, and making sure PWM is enabled in `/boot/config.txt`, googling for PWM channels on the Pi 5 seems to suggest the RP1 PWM controller uses `pwmchip2` on the Pi 5, while hardware PWM on previous Pi models was accessible through `pwmchip0`. You could verify this with `ls /sys/class/pwm` and check which devices are available. RPPAL is currently hardcoded to always use `pwmchip0`, so it's possible that code needs to be tweaked for the Pi 5.

If you have some time to check, grab a copy of the RPPAL repo, and in the `pwm::sysfs` module, try changing any `pwmchip0` paths to `pwmchip2` and see if that helps.

whattheflukz, 2024-04-05:
Hello @golemparts !
`ls /sys/class/pwm` reports `pwmchip0 pwmchip2 pwmchip6'
I tried cloning the repo and changing all instances of `pwmchip0`to `pwmchip2` inside of the `pwm::sysfs` module 
I also tried running the example code found here https://github.com/golemparts/rppal/blob/master/examples/pwm_servo.rs verbatim with both `pwmchip2` as well as `pwmchip0` set in `pwm::sysfs`. 
I also probed GPIO12, GPIO13, GPIO18, and GPIO19 for all tests just in case i was an idiot and just looking at the wrong pins

golemparts, 2024-04-05:
Thanks for trying `pwmchip2`. It's worth checking `pwmchip6` as well since, as far as I'm aware, it just uses the next available device number. However, if that also doesn't fix the issue, I'm not sure what else it could be. Let me know whether using `pwmchip6` resolves the issues. If not, I'll have to mark this as Help Wanted and get a hold of a Pi 5 myself so I can figure things out.

whattheflukz, 2024-04-07:
Hi again @golemparts 
I just tried pwmchip6 , also I did make sure to enable pwm in `/boot/firmware/config.txt` (as I'm using raspberry pi OS) I tried both `pwm` as well as `pwm-2chan` unfortunately no luck with any of this.

whattheflukz, 2024-04-07:
I also just tried doing the exact same setup on a raspberry pi 4, running the example code verbatim as well, and that works as expected so it's definitely a problem specific to the pi 5. I imagine because of the hardware differences with the new RP1 chip.

golemparts, 2024-04-09:
Thanks for the additional checks. Yeah, it definitely sounds like an RP1 specific issue. I'll have a Pi 5 coming in soon so I can figure out what's going on.

anthonyhoegberg, 2024-05-13:
Hello Im experiencing the same issue, anyone figured out any workaround to get pwm for rpi5 to work?

golemparts, 2024-05-13:
I finally have a Pi 5 available for testing, so I should be able to dive into this as soon as I get a chance.

RPPAL also has a software PWM available if you're going with @whattheflukz 's alternative suggestion. Worth a shot until this is resolved.

golemparts, 2024-05-17:
I did some testing today. GPIO18 on the Pi 5 is tied to pwmchip2/pwm2, while RPPAL is hardcoded to use pwmchip0 with pwm0/pwm1 on all Pi models. I assume this change is related to the RP1 now handling the hardware PWM channels. I'll add an exception for Pi models that have an RP1 (and also check the other hardware PWM channels), after which everything should be working as expected.

golemparts, 2024-05-17:
A fix has been implemented on the master branch. I've run some tests on both Pwm0 and Pwm1 on a Pi 4 and a Pi 5, and everything seems to be working. I'll schedule a 0.18.0 release for tonight or tomorrow, which will include the fix.

whattheflukz, 2024-05-18:
> A fix has been implemented on the master branch. I've run some tests on both Pwm0 and Pwm1 on a Pi 4 and a Pi 5, and everything seems to be working. I'll schedule a 0.18.0 release for tonight or tomorrow, which will include the fix.

Thanks so much for your effort!

golemparts, 2024-05-18:
Fixed with release 0.18.0

closed 2024-05-18 by golemparts