golemparts/rppal #146: the example `pwm_blinkled.rs` does not seem to work on my raspberry pi zero w

exo-cortex, 2024-05-21:
I tried out the example from the examples folder with no success. A python program that I found (here)[https://tutorials-raspberrypi.de/programmieren-lernen-raspberry-pi-gpio-pwm/] *does* work, though.
I also tried running the the example with sudo, with not success.

I use Raspberry Pi Os with the latest updates.

golemparts, 2024-05-21:
Based on the source code in the article you linked, it looks like the python program uses software PWM, which requires no additional setup. RPPAL also supports software PWM on a GPIO `OutputPin` with `set_pwm`, `set_pwm_frequency` and `clear_pwm`.

The `pwm_blinkled.rs` example uses hardware PWM, which is more accurate than software PWM. In order to use hardware PWM,  you'll first have to enable PWM in `/boot/firmware/config.txt` by adding `dtoverlay=pwm` and then rebooting your Pi. This will enable hardware PWM on GPIO 18 (physical pin 12).

exo-cortex, 2024-05-21:
Thank you for your reply! 
I did add `dtoverlay=pwm` to my `/boot/firmware/config.txt`, although I wasn't really sure where or if it is important where in the config file to put it. I also read somewhere that there are some udev-rules that I have to use. I don't remember where I found it, but I also tried that out. Is there something else I could try?

golemparts, 2024-05-21:
Adding it to the end of your config.txt should be fine. You can find the udev rules you mentioned [here](https://docs.golemparts.com/rppal/0.18.0/rppal/pwm/index.html), although they might not be needed anymore with the latest Raspberry Pi OS release. Either way, running the code using `sudo` would give you the same results.

Try running `pinctrl get 18` to make sure the pin is configured for PWM. If some other code changed it to an output pin (like that python example), that could be a possible reason why it's not working. Rebooting your Pi or manually setting it to the appropriate alt function should fix that. Also make sure you're using the correct pin (physical pin 12) for hardware PWM.

exo-cortex, 2024-05-21:
running `pinctrl get 18` gets me: 
`18: a5    -- | lo // GPIO18 = PWM0`
And I just realized that I completely misunderstood the pinmap provided from raspberry Pi. I just thought the pins were numbered weird, but they were in fact the microcontrollers GPIO-pins and NOT the raspberry Pi pins. 
I am very sorry for having wasted your time. But thank you nonetheless. I just managed to drive my first hardware-pwm :D

exo-cortex, 2024-05-21:
Closed because the issue was only my misunderstanding of the pinmap.

golemparts, 2024-05-21:
No problem. Glad to hear you figured it out! 😁