golemparts/rppal #147: Multiple slave select lines on one SPI bus

willAlbertini707, 2024-05-23:
I was reading through the rppal documentation and am wondering if a single Spi struct allows for multiple slave select pins. From what I can see there is only a single slave select line defined in new(). I cannot find a way to switch or change this value. Should I create a new Spi struct for each slave device? Any help would be appreciated, thanks!

golemparts, 2024-06-03:
Hey @willAlbertini707, apologies for the late reply. For SPI, RPPAL is just a wrapper for the underlying linux driver accessed through the available `spidev` character devices. Since each bus/slave select combo is accessed through a different `spidev` interface, you would have to create a new `Spi` struct for each slave device. I haven't actually experimented with this myself, but in theory that should work.

An alternative option would be to create an `Spi` struct for the SPI bus you want to use, and instead of using the pre-configured SS pins, use any other available GPIO pins for Slave Select and manually change those to high or low as needed.

willAlbertini707, 2024-06-06:
I created a new Spi struct for each and it seemed to work, thanks!

closed 2024-06-07 by golemparts