golemparts/rppal #148: Thanks for rppal, showcase of our app using it, and question

andrewdavidmackenzie, 2024-06-03:
Didn't know where to post this - so ended up using an issue. Sorry if no tthe best way.

1) Thanks for creating rppal. I've used in a couple of pi projects now and enjoying it.

2) New project of ours that uses it:

Two of us started a small project called “pigg” to learn more rust, Iced GUI and GPIO programming on the Pi with rppal.
The result is a first release (v0.1.0) that includes the “piggui” (“piggy” to friends) Iced GUI.

[image]

It runs on macOS/Linux/Windows hosts (mainly for dev for now, until we get remote GPIO implemented) and more importantly on a Raspberry Pi.
We hope this seems interesting, and potentially useful, especially when developed further. We’d love to hear your thoughts here, in GH issues or in GH discussions
Repo: https://github.com/andrewdavidmackenzie/pigg
Release Notes: https://github.com/andrewdavidmackenzie/pigg/releases/tag/0.1.0
[crates.io](http://crates.io/) entry: https://crates.io/crates/pigg
GH Issues: https://github.com/andrewdavidmackenzie/pigg/issues
GH Discussions: https://github.com/andrewdavidmackenzie/pigg/discussions
Release Discussion Thread: https://github.com/andrewdavidmackenzie/pigg/discussions/122

3) We have doubts about correct pin definitions, and what things are hardware and what sw. based on reading pi docs, rppal docs, pinout.xyz

I have listed the pins and doubts in this github issue:
https://github.com/andrewdavidmackenzie/pigg/issues/102

If there is any Pi expert here who can shed some light on the matter it would be much appreciated!

golemparts, 2024-06-03:
That's a nice UI and interesting project you have there! I'll have to play around with it.

As for the alternate function definitions, you'll probably want to use a different set for each set of Pis that shares the same SoC family. I'd go with the definitions as specified in the datasheets. So far I've avoided the issue by using generic `Alt0`-`Alt8` names, but I should probably consider adding the actual function names at some point as well. Also remember the Pi 5 has 3 additional Alt functions compared to previous models, and a pin's function can technically be NULL on the Pi 5 as well.

andrewdavidmackenzie, 2024-06-03:
Thanks!
And we hope "that's only the beginning"...

If you have suggestions please share them in issues or discussions.

I guess I'll have to slog through all the data sheets and compile per
device?

Your docs references to 3 I2C channels through us off...

Have you had any requests related to HATs?

andrewdavidmackenzie, 2024-06-03:
I was thinking. There is the python `pigpio` library, that can be run as the `pigpiod` daemon on the Pi, to effectively export it's functionalities over the network.

I was wondering if you'd thought about doing something similar with rppal?

We want to make the functionality available over the network, to a remote GUI.

`pigpiod` seems to focus on making it available to python libraries running on a macos/linux/window machine.

We are very happy using rppal, and having it available as a system-service on a Pi, accessible from a remote UI would be perfect.
Others could focus on "remote rppal" library and remote rust, or remote scripts if they wanted.

golemparts, 2024-06-05:
I haven't had any HAT-related requests, but RPPAL is only meant to support the base peripherals, so adding support for HATs seems more appropriate as a separate project.

Similarly, providing remote access doesn't really fit the scope I have in mind for RPPAL. I've considered it, but I think a separate project that uses RPPAL internally would be a better fit. I don't have any current plans on developing one though, so if you're up for it you're more than welcome to!

andrewdavidmackenzie, 2024-06-05:
Ok, that's clear. Thanks.

Pigpiod is a separate small project that creates a daemon out of the lib.
Something along those lines. "Piglet" may evolve into something along those
lines. If you want to, just follow the project and we'll see out it evolves.

Thanks again!

closed 2024-06-07 by golemparts