golemparts/rppal #153: UART. Losing chars on read

pahlavaubivca, 2024-08-18:
Hello! Got problem with UART reading.

[Repo with example](https://github.com/pahlavaubivca/rpi-backend-device/blob/feature/uart-read/src/main.rs)

I'm sending message like "len=29&wh=128,128&kc=l&keypressms=0\r\n" throught uart from raspberry pico (using rp2040_hal::uart).

But when i trying catch message on my raspberry pi - i losing bytes. For example on message above on pi a got (console output)

`UART read buf: [108]
UART read: "l"
UART read size: 1
UART read buf: [50]
UART read: "2"
UART read size: 1
UART read buf: [50]
UART read: "2"
UART read size: 1
UART read buf: [38]
UART read: "&"
UART read size: 1
UART read buf: [38]
UART read: "&"
UART read size: 1
UART read buf: [121]
UART read: "y"
UART read size: 1
UART read buf: [101]
UART read: "e"
UART read size: 1
UART read buf: [109]
UART read: "m"
UART read size: 1
UART read buf: [61]
UART read: "="
UART read size: 1
UART read buf: [10]
UART read: "\n"
`

On other pico, which i use as reader - message from above received without loses, but on rapsberry pi i almost every time i lost bytes...
Cable for commutation i use the same.

Maybe someone got the same problem and solve it?

Thanks!

golemparts, 2024-08-20:
Thanks for the report @pahlavaubivca. It's hard to say what the cause is of your problem. RPPAL is just a wrapper around the linux serial device drivers. While it's definitely possible there's some undiscovered bugs in either RPPAL or the drivers, it seems more likely the issue lies elsewhere.

Missing characters sounds like the devices aren't using the same speed, parity, data bits or stop bits. I would start with using the same low speed on both and make sure the other settings are the same as well. You could also experiment with different read modes, or read more than 1 byte at a time to see what kind of results you get.

golemparts, 2024-08-20:
Also make sure you're not flushing the incoming buffer on the Pi, or the outgoing buffer on the Pico, other than at the start of your code.

pahlavaubivca, 2024-08-20:
I found the problem. Main problem is my lacking of experience in work with bare metal, but also small configuration issues.

My setup 
1) Raspberry Pico which work as simple transmitter
2) Rapberry Pi 4/5 as receiver. 
OS
`
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
`

I'm using recommend setup via `/dev/ttyAMA0`

## Working setup.

1) /boot/firmware/cmdline.txt

`console=tty1 root=PARTUUID=<YOUR_UUID> rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=UA`

Here enough default to make manipulation with `sudo raspi-config` as any docs suggest.

2) /boot/firmare/config.txt
You need for sure have this 2 lines to be able work with ttyAMA0 and default uart GPIO (14/15), also as described in manual remove any lines with `enable_uart=`.
`
dtparam=uart0=on
dtoverlay=disable-bt
`
I tested on 2 Pi's (4/5) and both refuse to work without `dtparam=uart0=on`

3) Most annoying (for me) and maybe for someone else most obvious - is connect between your devices on only TX/RX line but also **you need common Ground**. That is main reason why i lost my bytes and its confused me a bit, because it still work without ground but very unpredictable, so here i spend massive amount of time, because almost all tutor don't write about this at all, as i understand for experienced users its obvious, but not for rookies as me... 

It shouldn't be problem if both devices receive power supply from the same source, but if your devices have independent power sources as individual batteries - common ground connection is must!

## Conclussion

What additional steps i needed which i not found in manuals is 
1) `dtparam=uart0=on` in config.txt file
2) common ground connection between my 2 devices, if you use distinctive power sources for your devices

Thanks for help and fast response! also maybe my experience same some time for other people)

golemparts, 2024-08-20:
Glad to hear you figured it out! Hopefully this'll help other people with similar serial issues.

closed 2024-08-20 by golemparts