golemparts/rppal #155: Support re-configuring pull mode on InputPins after construction

KronsyC, 2024-09-01:
I've come across a use case where it would be nice to support `set_mode` on the InputPin type.
It appears to already be implemented on `IOPin` so I don't see an issue with providing a similar interface for the `InputPin`.

I'd be happy to submit a PR for this

KronsyC, 2024-09-01:
Implementation could be pushed back to the impl_input! macro, as this is behavior i believe you'd want on any input-like pin

golemparts, 2024-09-04:
Thanks for the suggestion @KronsyC. While bias typically doesn't need to be changed once it's configured, I don't see a reason not to add `set_bias` to `InputPin` if there is a specific use-case that requires it. I should be able to include it in the next update.

closed 2024-09-05 by KronsyC