golemparts/rppal #161: PWM code is taking 'No such device' error.

system-linux, 2024-10-14:
Using this library with my Raspberry Pi 5, I ran the ready-made example code below using sudo:
`
use std::error::Error;
use std::thread;
use std::time::Duration;
use rppal::pwm::{Channel, Polarity, Pwm};

fn main() -> Result<(), Box<dyn Error>> {
    // Enable PWM channel 0 (BCM GPIO 18, physical pin 12) at 2 Hz with a 25% duty cycle.
    let pwm = Pwm::with_frequency(Channel::Pwm0, 2.0, 0.25, Polarity::Normal, true)?;
    // Sleep for 2 seconds while the LED blinks.
    thread::sleep(Duration::from_secs(2));
    // Reconfigure the PWM channel for an 8 Hz frequency, 50% duty cycle.
    pwm.set_frequency(8.0, 0.5)?;
    thread::sleep(Duration::from_secs(3));
    Ok(())
}
`
But this code does not work and gives the error `Error: Io(Os { code: 19, kind: Uncategorized, message: "No such device" })`.
Note: I ran the program with sudo due to an permission error.

golemparts, 2024-10-30:
Apologies for the late response. This type of error usually means PWM hasn't been properly configured. Make sure you have added `dtoverlay=pwm` to your `/boot/firmware/config.txt` (and reboot the Pi after).

golemparts, 2024-11-28:
@system-linux were you able to resolve this issue?

golemparts, 2024-12-10:
Closing this due to inactivity. If you're still encountering this error, and it's not a Raspberry Pi OS configuration issue, please let me know and we can revisit the matter.

system-linux, 2024-12-11:
I'm so sorry, I didn't have time to reply. I don't have time to try this anymore in my project because we decided to use C++ wiringPi. So I haven't had a chance to try it yet.