golemparts/rppal #162: Add support for MODE_NO_CS to SPI devices

dbartussek, 2024-10-18:
I am communicating with a device that needs a few milliseconds to register that CS has been asserted and must therefore manually control the CS pin.

In Python, this can be achieved with spidev by setting the `no_cs` member on an SpiDev instance to true and manually driving the CS pin. 
I've now tried to port this code to Rust but found that the [`MODE_NO_CS` flag exists as a constant](https://github.com/golemparts/rppal/blob/a614532f72642989194e25bfc5ca3073f9bd438b/src/spi/ioctl.rs#L59), but it isn't exposed.

golemparts, 2024-10-30:
When I initially wrote the SPI code, `SPI_NO_CS` wasn't fully supported by either the underlying drivers or hardware. See [this section](https://docs.golemparts.com/rppal/0.19.0/rppal/spi/index.html#not-supported) of the documentation for a possible workaround.

Since it's been a while since I last checked, it's possible support for NO_CS has been added in the mean time. On which Pi device were you able to get it to work in Python?

janderholm, 2025-06-28:
I don't know if `SPI_NO_CS` is implemented yet. But there is a devicetree overlays which sort of does the same.

From `/boot/overlays/README`:
```
Name:   spi0-0cs
Info:   Don't claim any CS pins for SPI0. Requires a device with its chip
        select permanently enabled, but frees a GPIO for e.g. a DPI display.
Load:   dtoverlay=spi0-0cs,<param>=<val>
Params: no_miso                 Don't claim and use the MISO pin (9), freeing
                                it for other uses.
```

closed 2025-07-01 by golemparts