golemparts/rppal #163: HW PWM fails to start if the pin has ever been setup on the pi5

Rylan-Meilutis, 2024-10-29:
After using the gpio pin for software pwm or other gpio operations, attempting to use the hardware pwm causes nothing to happen. This occurs because the pin config needs to be `18: a3    pd | lo // GPIO18 = PWM0_CHAN2`. This can be checked by running `pinctrl get 18`. Setting up a pin to hardware pwm does not setup the pin to that state.

golemparts, 2024-10-30:
Hi @Rylan-Meilutis. RPPAL uses the `pwm` sysfs interface to manage the hardware PWM channels. The interface is configured by enabling the appropriate overlays in `/boot/firmware/config.txt`, which will set the relevant GPIOs to the correct mode. RPPAL isn't directly involved in setting these, as it's not aware of which GPIOs are used for the selected PWM channel.

If you're changing the modes of the GPIOs the system has configured for hardware PWM, you'll have to change them back to the correct alternate mode as well. If you want to do this within RPPAL, you can set the correct alternate mode by using an `IoPin` and setting `reset_on_drop` to false so the mode isn't reset when the pin instance goes out of scope.

Rylan-Meilutis, 2024-11-01:
Thanks for the info. I ended up embedding the rpi utils pinctrl binary into my program to set the pin mode when needed.