golemparts/rppal #164: Permission denied on "/dev/gpiomem"

andrewdavidmackenzie, 2024-11-04:
This is a new error I am just starting to get:

```
thread 'tokio-runtime-worker' panicked at src/hardware_subscription.rs:150:22:
called `Result::unwrap()` on an `Err` value: Permission denied: /dev/gpiomem
```

The only thing I can think of that has changed is that I updated a lot of packages on Pi OS recently.

```
uname -a
Linux rpi400 6.6.58-v8+ #1809 SMP PREEMPT Wed Oct 23 11:53:53 BST 2024 aarch64 GNU/Linux
```

I'll try and track down the exact call and add it here.

andrewdavidmackenzie, 2024-11-04:
Confirmed that running as root avoids the problem.

golemparts, 2024-11-04:
Make sure `/dev/gpiomem` has `rw` permissions set for the `gpio` group, and the user you're logged in with is part of the `gpio` group. It looks like you may be running a different distro, which can cause issues. I suggest installing a fresh image of the latest stable Raspberry Pi OS release.

andrewdavidmackenzie, 2024-11-04:
I'll check those things and get back to you.

It worked fine before on this machine without changing any permissions..., with same distro/install. 

All I did is update all packages....maybe that changed something...

golemparts, 2024-11-28:
@andrewdavidmackenzie were you able to resolve this issue?

andrewdavidmackenzie, 2024-11-28:
No, I assume it is a distro issue...and in my case an update changed the permissions...

golemparts, 2024-11-28:
Out of curiosity, what distro are you running? Would be good to know in case this pops up again.

andrewdavidmackenzie, 2024-11-29:
Pi OS
```
uname -a
Linux rpi400 6.6.58-v8+ #1809 SMP PREEMPT Wed Oct 23 11:53:53 BST 2024 aarch64 GNU/Linux
```