golemparts/rppal #167: Failure to configure GPIO if run via systemd although no error propagated

zacharytomlinson, 2024-11-20:
I have had hits and misses with this thus far. I have gotten the GPIO configuration to work a few times if I manually start the application after all services have loaded. If run as a systemd service or via rc.local, no error is thrown but subsequent checking of the GPIO voltages indicates that either 1) they were never configured or 2) the configuration was set temporarily and then dropped after all services loaded. I had set the set_reload_on_drop(false) as well. It should be noted doing the same exact operations via python works every time when run via rc.local, I have not tried that one as a systemd service.

golemparts, 2024-11-21:
Hey @zacharytomlinson. I don't have experience interacting with the GPIO with an application that's managed by systemd, however my guess would be you'll have to configure `After=` to wait long enough for the relevant services to finish loading, as it sounds like something is overwriting the pin status. Something else to check in both cases is to make sure the user the process is running as has the correct permissions to access the GPIO devices. If it's run as root that shouldn't be an issue.

If your application works fine when run it directly from the command line, it does sound like either a configuration or permissions error. If the above-mentioned doesn't help, you'll probably find better Linux support on the official Raspberry Pi forums to help figure this out.

closed 2024-11-26 by zacharytomlinson

lislis, 2025-01-23:
hi and sorry for commenting on the closed ticket. @zacharytomlinson I'm running into the same issue, how did you fix it?

zacharytomlinson, 2025-01-23:
Unfortunately didn't use rppal. I either changed the device tree to set the default state of that pin low or I used a before clause in the systemd script that would turn the gpio low prior to the application being run.

lislis, 2025-01-24:
using before= in systemd script seems like a sensible approach, I'll try that, thanks!