golemparts/rppal #168: Allow parsing pwm::Channel from u8

tidely, 2024-12-09:
I'm currently working on a project where we calibrate individual PWM connections using software and hardware PWM. We would like to be able to parse the PWM Channel enum from a u8 or a comparable integer type. Currently the Channel types are represented as a `isize` integer, however there seems to be no way to convert it back. Could we maybe introduce a `TryFrom` trait for the `pwm::Channel` enum. Optionally there could be a `clap` feature which would add general parsing capabilities, but that seems rather excessive.

For example:

```rust
impl TryFrom<u8> for Channel {
    type Error = Error;

    fn try_from(value: u8) -> result::Result<Self, Self::Error> {
        match value {
            0 => Ok(Channel::Pwm0),
            1 => Ok(Channel::Pwm1)),
            _ => Err(Error)
        }
    }
}
```

mnmnk43434, 2024-12-09:
Hello, we tried to solve the issue.

This is what we did:

Added TryFrom<u8> implementation for the Channel enum to allow parsing from u8 values. This allows converting integer values to PWM channels, which is useful for calibrating individual PWM connections using software and hardware PWM.

You can review changes in this commit: https://github.com/mnmnk43434/golemparts-rppal-168/commit/de9d5c29266a5b023203e469e502f5166e4cbb9d.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

golemparts, 2024-12-09:
> Not sure if we should convert from 0 and 1 to Channels Pwm0 and Pwm1, or if we should convert Gpio pin numbers to their corresponding pwm channel.

RPPAL currently has no knowledge of which GPIOs the channels have been configured for, so converting them as channel IDs would be the easiest to implement.

tidely, 2024-12-09:
Thanks for your quick response! I'd be satisfied with any type of conversion, however I'd like to thrown in an idea, let me know if it could be a good fit.

Pins 12 and 18 always map to Pwm0, pins 13 and 19 always map to Pwm1. This means we could convert from the pin number to the Channel, but not the other way around.

I'll link the documentation [here](https://github.com/dotnet/iot/blob/main/Documentation/raspi-pwm.md#activating-2-channels) so you can find it easily.

golemparts, 2024-12-09:
On the Pi 5, and presumably all newer devices with an RP1 or equivalent, you can configure up to 4 PWM channels using any of the GPIOs, so even for the current models the pins could be different. See the `pwm-pio` overlay. I'm not sure how well supported it currently is though. I just added support for the two additional channels as I wasn't aware of this either.

commit b48c17a2 (2024-12-10): Add TryFrom<u8> implementation for Channel

Fixes #168
closed 2024-12-10 by golemparts