golemparts/rppal #171: I2C does not follow the `embedded_hal::i2c::I2c::transaction` contract

dimpolo, 2025-01-05:
Hi everyone. Could you have a look at bundling same operations in your I2C implementation?
This is required for the pn532 crate to work correctly, see https://github.com/WMT-GmbH/pn532/issues/27

Two consecutive reads should be bundled together.
Two consecutive writes should be bundled together.

Quote from the [docs ](https://docs.rs/embedded-hal/latest/embedded_hal/i2c/trait.I2c.html#tymethod.transaction)(emphasis mine):

Transaction contract:

- Before executing the first operation an ST is sent automatically. This is followed by SAD+R/W as appropriate.
- **Data from adjacent operations of the same type are sent after each other without an SP or SR.**
- Between adjacent operations of a different type an SR and SAD+R/W is sent.
- After executing the last operation an SP is sent automatically.
- If the last operation is a Read the master does not send an acknowledge for the last byte.
- ST = start condition
- SAD+R/W = slave address followed by bit 1 to indicate reading or 0 to indicate writing
- SR = repeated start condition
- SP = stop condition

dimpolo, 2025-01-05:
Related: https://github.com/rust-embedded/linux-embedded-hal/issues/82

GChalony, 2025-02-11:
Well I've had a look at the linux documentation

> Note that only a subset of the I2C and SMBus protocols can be achieved by
the means of read() and write() calls. In particular, **so-called combined
transactions** (mixing read and write messages in the same transaction)
**aren't supported**. For this reason, this interface is almost never used by
user-space programs.

So we cannot comply with the `embedded_hal`  spec if we're only reading / writing to the file. Maybe we could use [rust-i2cdev](https://github.com/rust-embedded/rust-i2cdev/blob/master/src/ffi.rs#L452) as a dependency ? Them seem to do it properly, and that's what is used by `linux-embedded-hal`

golemparts, 2025-02-11:
Thanks for your offer to work on this @GChalony!

> So we cannot comply with the `embedded_hal` spec if we're only reading / writing to the file. Maybe we could use [rust-i2cdev](https://github.com/rust-embedded/rust-i2cdev/blob/master/src/ffi.rs#L452) as a dependency ? Them seem to do it properly, and that's what is used by `linux-embedded-hal`

I prefer not to add any dependencies for core functionality, especially something this small. As rust-i2cdev uses the same underlying ioctl interface as RPPAL does, you should be able to implement the same solution using `REQ_RDRW`. Take a look at [ioctl.rs](https://github.com/golemparts/rppal/blob/master/src/i2c/ioctl.rs#L184C7-L184C15) and at [write_read](https://github.com/golemparts/rppal/blob/master/src/i2c.rs#L418) for an example on how it's used in a single write+read operation using this sequence:

START → Address + Write Bit → Outgoing Bytes → Repeated START → Address + Read Bit → Incoming Bytes → STOP

It's been a while since I last worked on that section of the code though. After skimming through I came across this note:

```rust
// NOTE: REQ_RDWR - Only a single read operation is supported as the final message (see i2c-bcm2835.c)
```

Granted, that note is over 7 years old, so things might have changed, but at least on older Pi devices the required functionality needed to implement a transaction with any number of reads or writes in any order isn't fully supported, and will return an error if there is more than a single read message, and the read isn't the final operation. See [i2c-bcm2835.c](https://github.com/raspberrypi/linux/blob/d664f45f77423b03e5a435e480822075c03331bd/drivers/i2c/busses/i2c-bcm2835.c#L457).

closed 2025-07-01 by golemparts